Fraunhofer-AISEC/gallia #140: strange race condition in scanners

rumpelsepp, 2022-05-30:
Sometimes the vECU tests fail:

* https://github.com/Fraunhofer-AISEC/gallia/runs/6649856371?check_suite_focus=true
* https://github.com/Fraunhofer-AISEC/gallia/runs/6649776374?check_suite_focus=true

Find out why this happens.

rumpelsepp, 2022-05-31:
@fkglr Could you have a look at the vECU? It seems to be happen quite often now… It is strange that it only happens for 3.10.

rumpelsepp, 2022-05-31:
I am a :monkey:. The problem why this fails is that the scanners should be able to deal with this. Also, the session changing logic is flawed, since the scanners must always start in the DefaultSession.

peckto, 2022-06-27:
Because we do not spawn a new vECU for every test-case, the vECU state (like session) is not properly cleared between two test-cases.
I'm not sure, if we want to enforce the default session at startup for all scanners. Maybe optionally. 
Or we reset the vECU state between test-cases.

peckto, 2022-06-27:
Still now sure, why this happens:
```
Traceback (most recent call last):
  File "/home/runner/work/gallia/gallia/src/gallia/udscan/core.py", line 455, in _tester_present_worker
    await self.transport.write(bytes([0x3E, 0x80]), tags=["IGNORE"])
  File "/home/runner/work/gallia/gallia/src/gallia/transports/tcp.py", line 93, in write
    assert self.reader is not None and self.writer is not None, assertion_str
AssertionError: bug: transport is not connected
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/home/runner/work/gallia/gallia/src/gallia/udscan/core.py", line 477, in _tester_present_worker
    await asyncio.sleep(1)
  File "/opt/hostedtoolcache/Python/3.10.4/x64/lib/python3.10/asyncio/tasks.py", line 605, in sleep
    return await future
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/home/runner/work/gallia/gallia/src/gallia/udscan/core.py", line 282, in _run
    await self.teardown(args)
  File "/home/runner/work/gallia/gallia/src/gallia/udscan/core.py", line 573, in teardown
    await self.tester_present_task
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/runner/work/gallia/gallia/src/gallia/cli.py", line 83, in main
    sys.exit(args.func())
  File "/home/runner/work/gallia/gallia/src/gallia/udscan/core.py", line 325, in run
    return asyncio.run(self._run(args))
  File "/opt/hostedtoolcache/Python/3.10.4/x64/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/opt/hostedtoolcache/Python/3.10.4/x64/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
asyncio.exceptions.CancelledError
```

closed 2022-06-30 by rumpelsepp