Fraunhofer-AISEC/gallia #221: Mixin class for artifacts dir

peckto, 2022-07-15:
Currently, only classes based on `Scanner` have an artifacts dir to store results.
As some, but not all gallia scripts need an artifacts dir, we do not want to push it in the `Script` class.

Instead, the idea is to outsource the artifacts dir functionality as a mixin class.
So every gallia script can optional inherit from it, if needed.

peckto, 2022-07-26:
Creating a mixin class for the artifacts dir seems not to be so trivial.
In order to create the artifacts dir, we need the following components:
* `--data-dir ` cli arg (in `add_class_parser()`)
* `artifacts_dir ` attribute (in `__init__()` )
* `prepare_artifactsdir` function, which gets the optional `data_dir` from cli
* calculate artifacts dir (in `run()`)

Especially, the dependency to the arg parser causes trouble, as only Scanner classes have access to the gallia arg parser. But the mixin itself should not be a Scanner class.

One alternative solution would be, to make the artifacts dir attribute optional and let the scanner choose, if it should be created or not. This has the disadvantage of additional assert statements for the type checker. A static typed solution would be better.

rumpelsepp, 2022-08-17:
Mixin stuff is implemented in #231.

closed 2022-08-18 by rumpelsepp