Fraunhofer-AISEC/gallia #453: vecu display broken

LeventCelik, 2023-10-24:
I am using Gallia 1.3.0 (installed with pip) with Python 3.10.12 on a Ubuntu 22.04.03 LTS 64-bit, GNOME 42.9.

Running a vecu through `gallia script vecu "isotp://can0?src_addr=0x009&dst_addr=0x001" rng` successfully creates a virtual ECU, but the display is broken as can be seen from the picture:
[image]

The same holds with the _tcp-lines_ option.

On a side note, the user guide to creating virtual ecus is outdated, and does not have enough information about vecu options.

rumpelsepp, 2023-10-24:
Could you try it again with `--no-volatile-info`?

ferdinandjarisch, 2023-10-24:
Yes, @rumpelsepp is completely right, this slipped through when introducing volatile-info!

I already had a commit that fixes it locally, but not yet pushed upstream.
It is cherry-picked in https://github.com/Fraunhofer-AISEC/gallia/pull/454 now.

LeventCelik, 2023-10-25:
Yes, that solves the issue. Thank you @rumpelsepp and @ferdinandjarisch! Thank you also for the speedy response.

Could you also help me with generating a virtual ECU with standardized subservice identifiers? For instance, using 0x10, SBF of 0x1 for default session, 0x3 for extended, etc. Can I do that, and if I can, which command shall I use?

rumpelsepp, 2023-10-25:
According to the issue just follow #454. For your question, I created #455. The author of the virtual ecu will take care of this. :)

closed 2023-10-25 by LeventCelik