Fraunhofer-AISEC/gallia #526: UDS Server's sessions as a part of the URI

mich41v4294, 2024-04-22:
Since most scanning operations are session-dependent, would it make sense to include the available sessions of the UDS server to be scanned inside the URI? Eg. `isotp://vcan0?is_fd=false&is_extended=false&src_addr=0x701&dst_addr=0x700&session=1,2,3` or similar (if it's better to have hex vs dec here). Then, the subsequent commands (eg. service scan etc.) will not need setting an argument and could get all the information required from the URI to scan all the available sessions. This might also serve as a first move towards fully automated scanning, which I feel lacking in this tool.
Please let me know your thoughts on this :)

rumpelsepp, 2024-04-30:
Fully automated scanning will be partially addressed by the 2.0 rework; there are a few ideas publicly documented here: https://github.com/Fraunhofer-AISEC/gallia/discussions/479. TL;DR: `gallia` will be fully usable as a python library (and cli tool as is) enabling more use cases and more fine grained custom scanners.

Your proposed idea from my point of view does not contribute to this goal, since it would introduce a layer violation and increase the complexity of the code base. The target string is transport layer only (e.g., ISO-TP, CAN, …) and by design should not be mixed with the upper layers like UDS.

mich41v4294, 2024-05-04:
Understood, thank you! :)

closed 2024-05-04 by mich41v4294