Fraunhofer-AISEC/gallia #529: Add session switching path for scanners

mich41v4294, 2024-04-24:
Currently, when launching scanning functionality on a session to which is unable to switch from session `0x01`, the tool just says "Unable to switch to session" and goes on. Could we implement functionality where you could pass the session switch path into the argument?

In this specific case, the switch to `0x02` is only possible from session `0x03`:

Eg. session scan result:
```
gallia.scan.sessions [result]: * Session 0x02
gallia.scan.sessions [result]:      via stack: 0x01->0x03
```

Proposed CLI argument:
`--session 1->3->2`

fkglr, 2024-04-24:
The current approach that is already implemented is to first scan for available sessions with the specific session scanner (`gallia scan uds sessions`). 
When using this with the db (`--db /path/to/db`) there is already functionality in place, which stores the transitions between sessions in the database.
When using one of the other scanners (e.g. `gallia scan uds services`) or actually any script also with the same database argument it will fetch those transitions from the databse and try to enter the sessions using these paths (after a failed try to switch directly).
This has the advantage that it works out of the box for any script by modifying the ECU's `set_session` functionality:
https://github.com/Fraunhofer-AISEC/gallia/blob/35d50d7fef2e44e255c57533d11036cf277c4b39/src/gallia/services/uds/ecu.py#L253-L265

This is of course only suitable for such simple cases where the only requirement is to switch to one or more sessions before entering the destined session.
An alternative, which allows to handle more complex preconditions is to define an own ECU and overwrite the `set_session_pre()` method:
https://github.com/Fraunhofer-AISEC/gallia/blob/35d50d7fef2e44e255c57533d11036cf277c4b39/src/gallia/services/uds/ecu.py#L105-L109
(it seems the doc is outdated for this one, but `level` is just the session to be entered, this function however does not need to call the change to this session but only make sure the preconditions are fulfilled, so that a subsequent session change will succeed)

peckto, 2024-05-07:
Is the issue solved with the explanation from @fkglr ?
Are there any open points?

mich41v4294, 2024-05-07:
All good from my side, I am yet to test this with the DB. Currently feeling like the lack of analyzer (#205) is currently the biggest blocker in the usability of the database functionality.

closed 2024-05-07 by mich41v4294